Seeed-Studio/Seeed_Arduino_CAN #28: Message filtering letting through unwanted ids

Elektromeka, 2016-03-30:
There is a problem with message filtering. In the example code set_mask_filter_recv
 With the filter and mask settings:

   CAN.init_Mask(0, 0, 0x3ff);        
   CAN.init_Mask(1, 0, 0x3ff);

   CAN.init_Filt(0, 0, 0x04); 
   CAN.init_Filt(1, 0, 0x05); 
   CAN.init_Filt(2, 0, 0x06); 
   CAN.init_Filt(3, 0, 0x07); 
   CAN.init_Filt(4, 0, 0x08); 
   CAN.init_Filt(5, 0, 0x09); 

 For some reason in addition to receiving ids 0x04~0x09 the code also receives 0x404~0x409

tonton81, 2016-04-09:
okay, there are lots of people who have this issue, and most of the time it's because the bus is busy with data, people recommend you hold reset for 5 seconds before you load the filters, however, I have found a solution that is a bit crude but DOES work as im switching from ALL traffic to 1x 29-bit canid, and then back to all traffic, dynamically, and without error.
There should be a note on this as i worked hard debugging and finding out the result by myself

if you create a for loop, and take your masks and filters above, and let it loop twice, you'll have more success changing them with less likely to let traffic slip through. But, even though, maybe it still might not set, I have set my for loop to set masks and filters 3x! and this has resulted in fixing this issue on 500 bitrate bus. This method does not require resetting the canbus shield OR the arduino which is awesome. 

Also, you should add the additional info for ISO 15765-4, to specify frame flow should be in readme not necessarily in the examples however, if you send a buf to the ecu requesting vin, or requesting 4 pids, ecu will only send you the first 5 bytes starting at buf[3] position, to get the remaining (and this includes reading via(09 02 pid) you MUST send a frame flow ex   unsigned char frame_flow = { 0x30, 00,00,00,00,00,00,00 } ( only once is needed) you will get 1 or 2 frames comming in with the remainder of the pid data or vin ascii (char). 
Tony

tonton81, 2016-04-09:
to read vin you'll need to save 3 buffers, concatenate them needed bytes together (exclude the first unneeded bytes from each), then snprintf or memcpy them to a new buffer to be able to print it as ascii

Elektromeka, 2016-05-18:
Wow, no idea how you figured that out but that's awesome work, thanks!

closed 2016-05-18 by Elektromeka