Seeed-Studio/Seeed_Arduino_CAN #35: Only one message id received

Yamakaky, 2016-11-01:
Just tried your shield + library, but I only get one message (seems related to motor on/off and motor starter), but nothing when I move the wheel or push the brakes. Any idea why?
Tested on BMW X3.

anthonywebb, 2016-11-03:
Are you doing any filtering, can you share your code?

Yamakaky, 2016-11-03:
https://github.com/Yamakaky/can-bus
It's the example modified to output json.

Yamakaky, 2016-11-04:
Any idea?

anthonywebb, 2016-11-05:
Sorry, I do not.  Mine is working fine and your code looks great.

wurstkaiser, 2017-01-22:
Hey Guys, 

same problem to me :( Any solution for this?

Thanks

Yamakaky, 2017-01-22:
Apparently German cars are not great for using the CAN bus. American and Japanese ones seems better.

wurstkaiser, 2017-01-24:
thx for reply ;) but German cars are the better ones xD that for it must be possible to get this work.

BlueNova4001, 2017-01-24:
CAN warriors,
I haven't used the shield yet in testing (I'm using a more robust tool).

That said please understand a couple limitations. First,  most arduinos
aren't clocked fast enough to handle high speed CAN's max baud rate. Also,
cars prior to 2008 in the US market were not required to expose their CAN
bus over the OBD connection. Finally, each manufacturer complies slightly
differently. Just FYI.

sjonkeesse, 2019-08-21:
I got the same issue. When sending with 500 kbps, the shield sends only one of my 5 messages. When I'm lower the bau rate to 250 kpbs it does send all the messages.

`
unsigned char stmp11601[8] = {0, 1, 0, 0, 0, 0, 0, 0};
unsigned char stmp11602[8] = {1, 13, 0, 0, 0, 0, 0, 0};
unsigned char stmp11603[8] = {2, 0, 0, 132, 112, 0, 0, 0};
unsigned char stmp11604[8] = {3, 0, 0, 0, 0, 0, 0, 0};
unsigned char stmp11605[8] = {7, 0, 36, 0, 0, 0, 0, 0};

void loop()
{
    CAN.sendMsgBuf(0x1A0, 0, 8, stmp11601);
    delay(5);
    CAN.sendMsgBuf(0x1A0, 0, 8, stmp11602);
    delay(5);
    CAN.sendMsgBuf(0x1A0, 0, 8, stmp11603);
    delay(5);
    CAN.sendMsgBuf(0x1A0, 0, 8, stmp11604);
    delay(5);
    CAN.sendMsgBuf(0x1A0, 0, 8, stmp11605);
    delay(5);
}
`

`CAN.begin(CAN_500KBPS)`
[image]

`CAN.begin(CAN_250KBPS)`
[image]

closed 2020-12-10 by Pillar1989