Seeed-Studio/Seeed_Arduino_CAN #38: Init fail when USB host board initialized

timtt, 2016-12-19:
I'm running a USB host shield on top of the Seeed CAN board. If I don't initialize the USB shield, CAN works great. If I init the USB shield second, after a successful init of the CAN shield, I get a silent failure of the CAN shield where sending of CAN messages doesn't work. Receive works, but no sending of messages in that case.

If I init the USB shield first, then I get a CAN shield init failure in mcp2515_setCANCTRL_Mode(const INT8U newmode) where it can't set the newmode. The reading of the register doesn't return newmode and thus it fails.

Any thoughts on what's going on here and potential solutions to try? Is this more of a USB host driver question?

tonton81, 2016-12-20:
whats the model of your usb shield? sparkfun? anyways, according to a random usb host shield i looked online, it uses default CS as pin 10, if your CAN shield is using CS 10, then thats why.

timtt, 2016-12-20:
Thanks. I have two shields, the official one and the Keyes one. They both use pins 10 and 9, so I rerouted the CAN CS to 7, both physically and when initializing the library. Works great with just the CAN shield installed. Whenever either USB board is present and the USB driver is initialized, the CAN driver can't write to the register (0x0F) that's needed to set the CANCTRL mode.

I also made sure to disable the SPI bus on pins 11-13 by disconnecting them. Both boards therefore are operating SPI through the ICSP header.

I'm going to try the CAN board on a Duo and use it's onboard USB. Hopefully that works. For my project I need CAN and USB host capability.

tonton81, 2016-12-20:
no, 11, 12, and 13 is shared with both shields, keep them hooked up, also make sure your unused CS pins are held high with pinMode AND pullup of 10k

all 7 9 and 10 :)

tonton81, 2016-12-20:
if you dont set pinmode youll have that issue as well

timtt, 2016-12-21:
I figured out the problem. The official Arduino USB shield has a conflict that I cannot figure out how to resolve. As soon as it's attached, the CAN Bus shield is not reachable. I thought the same was happening with the Keyes USB board, but it turns out the issue was totally different. That shield has a relatively short ICSP female header and it wasn't making contact with the male part of the CAN Bus ICSP header (it slid in, but not far enough to make electrical contact). Once I cut down all the USB shield's male pins by 2 mm, the ICSP made contact and all is good now in being able to talk to both boards at the same time.

I did have to make sure to route pins 7, 9 and 10 only to the board which needs them.

closed 2017-05-25 by loovee