Seeed-Studio/Seeed_Arduino_CAN #39: Receiveing extended Identifiers

rwilso32, 2017-01-20:
I have merge the send and receive check files. I am sending a extended ID CAN message without any problems. When I try to receive an extended ID it seems to ignore the message? Any help would be great, here is the code:

#include <SPI.h>
#include "mcp_can.h"

// the cs pin of the version after v1.1 is default to D9
// v0.9b and v1.0 is default D10
const int SPI_CS_PIN = 10;

MCP_CAN CAN(SPI_CS_PIN);                                    // Set CS pin

void setup()
{
    Serial.begin(115200);

    while (CAN_OK != CAN.begin(CAN_500KBPS))              // init can bus : baudrate = 500k
    {
        Serial.println("CAN BUS Shield init fail");
        Serial.println(" Init CAN BUS Shield again");
        delay(100);
    }
    Serial.println("CAN BUS Shield init ok!");
}

unsigned char stmp[8] = {0x03,0x22,0x70,0x00,0x00,0x00,0x00,0x00};

void loop()
{
    // send data:  id = 0x00000000, extended ID, standard frame, data len = 8, stmp: data buf
    CAN.sendMsgBuf(0x18DA60F1, 1, 8, stmp);
    delay(100);                       // send data per 100ms

    unsigned char len = 0;
    unsigned char buf[8];

    if(CAN_MSGAVAIL == CAN.checkReceive())            // check if data coming
      {
        CAN.readMsgBuf(&len, buf);    // read data,  len: data length, buf: data buf
        
        Serial.print("ID:");
        Serial.print(CAN.getCanId(),HEX);
        Serial.print("\, ");
        for(int i = 0; i<len; i++)    // print the data
        {
            Serial.print(buf[i],HEX);
            Serial.print("\, ");
        }
        Serial.println();
     }
}

/*********************************************************************************************************
  END FILE
*********************************************************************************************************/

loovee, 2017-09-18:
Hello, do you till have this issue?
Thanks.

mikko80, 2018-01-11:
Hi! I have same problem too.

talkingbeast, 2018-02-11:
i met the same issue when i port this driver to esp8266

pepsifan92, 2018-02-11:
If I change the Line 31 from the sendBlink.ino example 
`CAN.sendMsgBuf(0x70,0, 8, stmp);` 
to the extended format:
`CAN.sendMsgBuf(0x70,1, 8, stmp);`
or with using an real "extended" identifier:
`CAN.sendMsgBuf(0x7000,1, 8, stmp);`

... The receiver of the message gets only CAN.checkReceive() == CAN_NOMSG.

So the condition 
`if(CAN_MSGAVAIL == CAN.checkReceive()) {...}`
never gets true and the extended format message can't get received. How to tell the MCP to accept messages in extended format?
(I've used the receive_check.ino example.)

An full example of communication with extended format would be great ;)

orionracing, 2018-02-12:
//demo: CAN-BUS Shield, receive data
#include <mcp_can.h>
#include <SPI.h>
#include <stdio.h>
#define INT8U unsigned char
#define INT32U unsigned char

//INT8U Flag_Recv = 0;
INT8U len = 0;
INT8U buf[8];
INT32U canId = 0x000;
INT8U len1 = 0;
INT8U buf1[8];
INT32U canId1 = 0x000;
const int SPI_CS_PIN=10;
MCP_CAN CAN(SPI_CS_PIN);

void setup()
{
  CAN.begin(CAN_500KBPS);                   // init can bus : baudrate =
500k
 // attachInterrupt(0, MCP2515_ISR, FALLING); // start interrupt
  Serial.begin(115200);
}
/*
void MCP2515_ISR()
{
     Flag_Recv = 1;
}*/

void loop()
{
   // if(Flag_Recv)                   // check if data was recieved
   // {
     // Flag_Recv = 0;                // clear flag
      CAN.readMsgBuf(&len, buf);    // read data,  len: data length, buf:
data buf
      canId = CAN.getCanId();
      //Print data to the serial console
      Serial.println("CAN_BUS GET DATA!");
      Serial.print("CAN ID: ");
      Serial.println(canId);
      Serial.print("data len = ");
      Serial.println(len);
      //This loops through each byte of data and prints it
      for(int i = 0; i<len; i++)    // print the data
      {
        Serial.print(buf[i]);
        Serial.print("\t");
      }
      Serial.println();
      delay(1000);
      CAN.readMsgBuf(&len1, buf1);    // read data,  len: data length, buf:
data buf
      canId1 = CAN.getCanId();
      //Print data to the serial console
      Serial.println("CAN_BUS GET DATA1!");
      Serial.print("CAN ID1: ");
      Serial.println(canId);
      Serial.print("data len1 = ");
      Serial.println(len1);
      //This loops through each byte of data and prints it
      for(int i = 0; i<len1; i++)    // print the data
      {
        Serial.print(buf1[i]);
        Serial.print("\t");
      }
      Serial.println();
      delay(1000);
   // }
}

*Try this code*

pepsifan92, 2018-02-12:
Hi,
that doesn't work either. Simply skip the check of available messages isn't the solution and I wonder how to configure the MCP2515 to accept the extended format Identifier. With standard Format everything works fine.

gggplaya, 2018-09-27:
I was able to fix this by going into the mcp_can.cpp file and changing ext to 1, it was set to 0 by default.  
```
void MCP_CAN::mcp2515_initCANBuffers(void)
{
    INT8U i, a1, a2, a3;
                   
    INT8U ext = 1;
    INT32U ulMask = 0x00, ulFilt = 0x00;
```

closed 2020-12-10 by Pillar1989

dhopf, 2021-04-20:
Was there some regression regarding this? I saw the old pull request to support receiving extended CAN identifiers. However, I still have the very same issue: I have two CAN shields with a Teensy 3.2.

- Sending from one and receiving on the other works fine with standard mode (11-Bit identifiers)
- Doing the same with extended 29-Bit identifiers: Frames are sent, but never received.

Extract from "sendReceive" example (adapted for MCP2515 with 8MHz crystal here) - **WORKING** with **normal** frames:
```
    CAN_SEND.sendMsgBuf(0x123, 0, 8, stmp); // NOTE: second parameter '0' specifies 'send normal frame'
    delay(100);                       // send data per 100ms
    SERIAL.println("CAN BUS sendMsgBuf ok!");

    // ---------------------

    if (CAN_MSGAVAIL == CAN_RECEIVE.checkReceive()) {
    // read data,  len: data length, buf: data buf
        SERIAL.println("checkReceive");
        CAN_RECEIVE.readMsgBuf(&len, buf);
        // print the data
        for (int i = 0; i < len; i++) {
            SERIAL.print(buf[i]); SERIAL.print(" ");
        }
        SERIAL.println();
    }
```

**NOT WORKING** with **extended** frames (CAN_RECEIVE.checkReceive() never returns CAN_MSGAVAIL):
```
    CAN_SEND.sendMsgBuf(0x123, 1, 8, stmp); // NOTE: second parameter '1' specifies 'send extended frame'
    delay(100);                       // send data per 100ms
    SERIAL.println("CAN BUS sendMsgBuf ok!");

    // ---------------------

    if (CAN_MSGAVAIL == CAN_RECEIVE.checkReceive()) {
    // read data,  len: data length, buf: data buf
        SERIAL.println("checkReceive");
        CAN_RECEIVE.readMsgBuf(&len, buf);
        // print the data
        for (int i = 0; i < len; i++) {
            SERIAL.print(buf[i]); SERIAL.print(" ");
        }
        SERIAL.println();
    }
```

I use the most recent version of this library.

"Changing" ext to 1 inside the mcp2515_initCANBuffers function is not an option since it seems to have been completely reworked in the meantime and I cannot find a suitable spot to "hack" this in:
```
void mcp2515_can::mcp2515_initCANBuffers(void) {
    byte i, a1, a2, a3;

    a1 = MCP_TXB0CTRL;
    a2 = MCP_TXB1CTRL;
    a3 = MCP_TXB2CTRL;
    for (i = 0; i < 14; i++) {                       // in-buffer loop
        mcp2515_setRegister(a1, 0);
        mcp2515_setRegister(a2, 0);
        mcp2515_setRegister(a3, 0);
        a1++;
        a2++;
        a3++;
    }
    mcp2515_setRegister(MCP_RXB0CTRL, 0);
    mcp2515_setRegister(MCP_RXB1CTRL, 0);
}
```

I appreciate all the effort which went into making this great library, saving hundrets of hours for people out there - I just begin to question my sanity after spending >>10h debugging this and still not understanding why I don't receive extended frames.

gggplaya, 2021-04-20:
> Was there some regression regarding this? I saw the old pull request to support receiving extended CAN identifiers. However, I still have the very same issue: I have two CAN shields (not the original Seeed ones, but also featuring the MCP2515, an NXP transceiver and a 8MHz crystal) with a Teensy 3.2.
> 
> Extract from "sendReceive" example (adapted for MCP2515 with 8MHz crystal here) - **WORKING** with **normal** frames:
> 
> ```
>     CAN_SEND.sendMsgBuf(0x123, 0, 8, stmp); // NOTE: second parameter '0' specifies 'send normal frame'
>     delay(100);                       // send data per 100ms
>     SERIAL.println("CAN BUS sendMsgBuf ok!");
> 
>     // ---------------------
> 
>     if (CAN_MSGAVAIL == CAN_RECEIVE.checkReceive()) {
>     // read data,  len: data length, buf: data buf
>         SERIAL.println("checkReceive");
>         CAN_RECEIVE.readMsgBuf(&len, buf);
>         // print the data
>         for (int i = 0; i < len; i++) {
>             SERIAL.print(buf[i]); SERIAL.print(" ");
>         }
>         SERIAL.println();
>     }
> ```
> 
> **NOT WORKING** with **extended** frames (CAN_RECEIVE.checkReceive() never returns CAN_MSGAVAIL):
> 
> ```
>     CAN_SEND.sendMsgBuf(0x123, 1, 8, stmp); // NOTE: second parameter '1' specifies 'send extended frame'
>     delay(100);                       // send data per 100ms
>     SERIAL.println("CAN BUS sendMsgBuf ok!");
> 
>     // ---------------------
> 
>     if (CAN_MSGAVAIL == CAN_RECEIVE.checkReceive()) {
>     // read data,  len: data length, buf: data buf
>         SERIAL.println("checkReceive");
>         CAN_RECEIVE.readMsgBuf(&len, buf);
>         // print the data
>         for (int i = 0; i < len; i++) {
>             SERIAL.print(buf[i]); SERIAL.print(" ");
>         }
>         SERIAL.println();
>     }
> ```
> 
> I use the most recent version of this library.
> 
> "Changing" ext to 1 inside the mcp2515_initCANBuffers function is not an option since it seems to have been completely reworked in the meantime and I cannot find a suitable spot to "hack" this in:
> 
> ```
> void mcp2515_can::mcp2515_initCANBuffers(void) {
>     byte i, a1, a2, a3;
> 
>     a1 = MCP_TXB0CTRL;
>     a2 = MCP_TXB1CTRL;
>     a3 = MCP_TXB2CTRL;
>     for (i = 0; i < 14; i++) {                       // in-buffer loop
>         mcp2515_setRegister(a1, 0);
>         mcp2515_setRegister(a2, 0);
>         mcp2515_setRegister(a3, 0);
>         a1++;
>         a2++;
>         a3++;
>     }
>     mcp2515_setRegister(MCP_RXB0CTRL, 0);
>     mcp2515_setRegister(MCP_RXB1CTRL, 0);
> }
> ```
> 
> I appreciate all the effort which went into making this great library, saving hundrets of hours for people out there - I just begin to question my sanity after spending >>10h debugging this and still not understanding why I don't receive extended frames.

Looks like that answered here, they put it in the sendFD instead:  https://github.com/Seeed-Studio/Seeed_Arduino_CAN/issues/114

dhopf, 2021-04-20:
Thanks for the reference @gggplaya, but I don't fully understand the correlation to this issue here. There, someone asked about setting the ext as well as the rtr bit. The ext bit can be set directly in the normal sendMsgBuf function - the rtr bit is automatically set to 0 in that case (which should be ok for my case). The overlaoded sendMsgBuf function is only needed if someone actually wants to set the RTR bit to not send data, but request data from another node.

Looking at the call hierarchy, it seems that the ext parameter is indeed used.

I wonder if the issue is on the sending or receiving end: Does the extended frame get sent out, but the receiver somehow doesn't understand it, or doesn't it even get sent?

gggplaya, 2021-04-20:
Send me your email address and I'll send you the older libraries.

> uest data from another node.
> 
> Looking at the call hierarchy, it seems that t

Send me your email and I'll send you the older library.

dhopf, 2021-04-20:
I cannot find any means to send it to you. I (obviously) don't want to put it here in a public discussion.
Maybe you can write me on Twitter @DanielHopf5 or give me some other means to contact you privately? Thanks upfront!

gggplaya, 2021-04-21:
> I cannot find any means to send it to you. I (obviously) don't want to put it here in a public discussion.
> Maybe you can write me on Twitter @DanielHopf5 or give me some other means to contact you privately? Thanks upfront!

I made a repository and invited you to  access it.

dhopf, 2021-04-21:
Thank you. Now I understand why I couldn't find similar code in that function neither from the most recent commits nor from the commits around your posting date (Sep. 2018): The version you referred to is v0.1 from 2012!
I will give it a try, but I really cannot believe that this bug has been in the lib ever since for 9 years now and no one noticed. I will investigate further and in case reopen a new issue for this.

gggplaya, 2021-04-21:
I'm not sure why the library was reworked, or what changes there were.  But I've been using this with their shield and it works fine.  No problems that I've encountered.  We've purchased a few arduino's and CAN shields in the last few years, so it's probably not due to board revisions.