Seeed-Studio/Seeed_Arduino_CAN #40: One Shield, Sending & Receiving at "same time" possible?

wurstkaiser, 2017-01-22:
Hi Guys, 
just have a question using the CAN-Shield with an Arduino uno.
For now i know that it is possible to send messages (which already works fine) to control two electric motors. 
But when i try to read ALL the messages which the electric motors are sending, most of them (about 15 out of 20) disappear and won't bei displayed in the Console.
But i know from another CAN-sniffer, that these messages definitely were sent. 
I tried this with the code from loovee receive_check.

Thanks for reply! And the great work on the code provided.
Wurstkaiser

loovee, 2017-02-01:
Hello, I guess this is because Serial.println need time to print things on the monitor, which cause the miss of message. I suggest that you make a larger buff and save all data and print in one time.
Thanks.

Loovee

wurstkaiser, 2017-02-06:
Thank you very much for that hint, i'll try this and come back to that. I'm not sure if that will work for me.

Sincerly,
wurstkaiser

closed 2017-05-25 by loovee