Seeed-Studio/Seeed_Arduino_CAN #41: Communication between Arduino UNO & Can-Shield

wurstkaiser, 2017-01-24:
Hello World,
i'm really new to all this stuff, but having lots of fun with that and try to understand. So here's my question, which surely can easily be answered by one of you :)
I'm trying to build up a sketch of the "wiring" of my system-layout, which is no Problem so far.
But actually i'm asking myself, which Pins are used for the communication between the CAN-Shield and an Arduino UNO? 
already looked at
https://github.com/SeeedDocument/CAN_BUS_Shield/raw/master/resource/CAN-BUS_Shield_v1.2.pdf
but still not sure, which pins are responsible for the communication?
Referring to the SPI, i think this are MOSI/MISO/SCK/GND on the ICSP terminal on the shield? Am I wrong?

Much Thanks for replying!

kind regards
wurstkaiser

loovee, 2017-02-01:
Hi Wurstkaiser,

Pin MOSI/MISO/SCK and D2(interrupt) it needed. And for Arduino, the pins for SPI is connected to ICSP as well.
Thanks. 

Loovee

closed 2017-02-01 by loovee

wurstkaiser, 2017-02-02:
Thanks a lot!