Seeed-Studio/Seeed_Arduino_CAN #53: Baud Rate Config

mcforcey, 2017-10-11:
Hello,
I have a baud rate that doesn't seem to be an option in the calculator software or other standard tables online.  It's a somewhat older system, and the only number I have to go off is 9600 - the value of a setting named Baud Rate in the main software.  Any guidance would be much appreciated.
Thank you.

wdross, 2017-10-11:
I suspect you are not looking at a CAN device, but instead an RS-232 serial device.

mcforcey, 2017-10-11:
I have one device connected to another using a 9-pin D connector, with 3 pins present - CAN High, CAN Low, and ground.  The receiving device is the one with a UI and that's where it has a setting for Baud Rate which is required to be set to 9600 for this implementation.  I just don't know exactly what that setting is referring to.  I'd like to be able to split the connection and read the messages using the CAN shield.

closed 2020-12-10 by Pillar1989