Seeed-Studio/Seeed_Arduino_CAN #67: 1Mbit - really?

vinnitu, 2018-07-06:
have stm32f103 with tfj1050 - and I can communicate with rasp 

oscilator=16000000
bitrate 1000000 in /etc/network/interfaces
but it is not 1Mb! It is more like ~140Kb
why?

now I try connect arduino uno + mcp2515 + mcp2551 + 8000 oscilator via spi 1Mb - and it doesnt work...

If I change bitrate to 500Kb - it works (< 500 ofcause) but I need 1Mb - what i need to do? is it possible in general?

cawena, 2018-08-02:
I have an 16000 oscilator and setup with CAN_500KBPS. When I check with CANalystII the baud rate is 250k bps and not 500.
Some thing is wrong here?

jerryyip, 2018-08-06:
@vinnitu  What is tfj1050? And which can-bus module you used on pi? Can you provide your testing method? Thanks.

jerryyip, 2018-08-06:
@cawena oscillator has nothing to do with the can-bus speed. Can you provide more information?

Oh-Doo-Yong, 2018-08-07:
I think...
//-----------------------------------------------------------------------------------------------
MCP_CAN MCP_CAN0;//case 16Mhz oscilator 
MCP_CAN MCP_CAN1;//case 8Mhz oscilator 

void setup()
{
MCP_CAN0.begin( CAN_500KBPS  );
//default 16Mhz 
// byte begin( byte speedset, const byte clockset = MCP_16MHz ); <--- mcp_can.h

MCP_CAN1.begin( CAN_500KBPS,  MCP_8MHz );//need  MCP_8MHz  <--- mcp_can_dfs.h
}
//-----------------------------------------------------------------------------------------------

cawena, 2018-08-23:
The init of the can bus is like that: 
while (CAN_OK != CAN.begin(CAN_500KBPS)) {}
The settings in the CANalyst-II for the CANBus are 250kbps, so that I can receive data.

vivekkairi, 2018-12-22:
@cawena Any updates?
Cuase even I'm facing half speed issue.

greghol, 2020-04-05:
For the people having speed issues it is because the examples are not correct when setting bit rate. The crystal freq param needs to be used in CAN.begin(). see implementation of the call. See #93

greghol, 2020-04-07:
> 
> 
> @cawena oscillator has nothing to do with the can-bus speed. Can you provide more information?

The crystal osc does have something to do with the can bus speed. How does it transmit and how does it sync up with incoming packet and pick the sample points for each bit? The examples are not using CAN.begin() correctly. See #93.

closed 2020-12-10 by Pillar1989