Seeed-Studio/Seeed_Arduino_CAN #79: CAN_83K3BPS for 8MHz not defined

dartrax, 2019-02-18:
` CAN.begin(CAN_83k3BPS, MCP_8MHz);`
won't work, because in mcp_can.cpp, there is no 83k3BPS case in the select statement for MCP_8MHz.
Also, in mcp_can_dfs.h, 
```
#define MCP_8MHz_83k3BPS_CFG1 (0xxx)
#define MCP_8MHz_83k3BPS_CFG2 (0xxx)
#define MCP_8MHz_83k3BPS_CFG3 (0xxx)
```
is missing.

Jenkinlu001, 2019-07-02:
Thanks for your suggestion, we will add it.

ETACSDecoder, 2020-06-26:
or
#define MCP_8MHz_83k3BPS_CFG1 (0x01)
#define MCP_8MHz_83k3BPS_CFG2 (0xbe)
#define MCP_8MHz_83k3BPS_CFG3 (0x07)

Pillar1989, 2020-12-10:
If you feel you need to continue the discussion, you can reopen the issues.

closed 2020-12-10 by Pillar1989

baorepo, 2026-04-08:
I opened PR #156 to add the missing MCP2515 8MHz + 83.3kbps support and included a regression test.

@dartrax @J-Dunn If either of you still has hardware using CAN.begin(CAN_83k3BPS, MCP_8MHz);, could you please test this branch/PR and report whether initialization and bus traffic behave correctly on your setup?