Seeed-Studio/Seeed_Arduino_CAN #82: Can Connection

mat-lol-mat, 2019-08-02:
I am stuck in a loop with the CAN Bus Init fail. I tried a few of the baud rate options (not all of them)
but I still seem to be stuck with this error
[image]

mat-lol-mat, 2019-08-02:
Currently I am trying to implement :
[image]

mat-lol-mat, 2019-08-02:
Currently, I am trying to implement:
[image]

mat-lol-mat, 2019-08-03:
Figured out my initial issue, I had the wrong PIN. However, I am still stuck on not being able to read my vehicle information

ICeZer0, 2019-10-06:
@markpwoodward I am getting the same issue running the HelloWorld CAN-BUS code. I have the shield and Arduino connected.  Is your baudrate set to 500k?

ICeZer0, 2019-10-06:
Fixed. Simply set SPI_CS_PIN = 9 instead of 10 for the input param MCP_CAN CAN(SPI_CS_PIN);
​

maqeelqureshi, 2020-03-21:
I am using SunFlower CAN Shield V3. https://www.aliexpress.com/item/32948283535.html
Setting CS to 10 resolves this error for me (default is 9 in example). 
But I don't receive any data.

GunZi200, 2020-07-23:
Seems there is no issue here. Shouldn't this issue be closed?

Pillar1989, 2020-12-10:
If you feel you need to continue the discussion, you can reopen the issues.

closed 2020-12-10 by Pillar1989

Aniltudarmstadt, 2022-01-11:
Hey ,
im using the adruino board von Joy-it to calibrate the H2 sensor but it always shows "can init failed"

09:20:19.968 -> CAN init fail, retry...
09:20:20.262 -> CAN init fail, retry...
09:20:20.585 -> CAN init fail, retry...
09:20:20.909 -> CAN init fail, retry...
09:20:21.244 -> CAN init fail, retry...
09:20:21.568 -> CAN init fail, retry...
09:20:21.891 -> CAN init fail, retry...
09:20:22.214 -> CAN init fail, retry...
 how to solve this problem ?
can any one help me?

Khalid-Hersafril-Flowcopter, 2023-03-08:
in my case, spi_cs_pin 10 works for me instead of 9