Seeed-Studio/Seeed_Arduino_CAN #94: Data transfer size  (unsigned char buf[8]) 0

Gbouna, 2020-04-29:
Thank you for this library, it is indeed very helpful. 
I noticed the buffer size is only 8 byte. My application involves sending a 6 by 6 array of data and each member of the array value could be as high as 500. How do i implement this, how do i send this on the CAN bus. A simple line of code or any idea will be well appreciated.
Thank you.

wdross, 2020-05-01:
CAN messages are physically limited to 8 bytes each.
With a value up to 500, you need two bytes for each value.
So 6 x 6 x 2 = 72 bytes of data would take a minimum of 9 messages.
You can implement some protocol of your own to keep track of which message within the sequence of messages your software is at, or you can look to implement something pre-existing.  The first thing that comes to mind is using CANOpen's definition of SDO (Service Data Objects) that define variables on a CAN node, and using this pre-existing protocol would result in sending 36 messages to transfer 6 x 6 elements.  This protocol already would keep track of which value is being updated as part of the message being sent.
https://en.wikipedia.org/wiki/CANopen#Service_Data_Object_(SDO)_protocol

Gbouna, 2020-05-07:
Thank you for your response, i will look into that.

closed 2020-12-10 by Pillar1989