Seeed-Studio/Seeed_Arduino_CAN #98: When in interrupt mode, the data coming can't be too fast, must >20ms, or else you can use check mode

zentT, 2020-07-17:
It's unclear what it means, can someone actually explains me. 

Also, why is that ? 

Thanks

zentT, 2020-07-20:
I'm planning to use this library in interrupt mode, on a 1mbps CAN bus. I'm sending short CAN messages with 4 bytes data. 
Is it possible ? I don't understand this warning "When in interrupt mode, the data coming can't be too fast, must >20ms, or else you can use check mode"

Thanks

Teddyz, 2020-07-24:
I searched for the text and found some code on dfrobot.com, with the statment in the title 
That code sets the variable `flagRecv` to high in the interrupt (yes, they should be short), and then still reads the received frame from the loop().

Now I am just guessing. That code has the line `unsigned char flagRecv = 0;`. Since the variable `flagRecv` is changed in the interrupt, it should be declared as volatile `volatile unsigned char flagRecv = 0;` (or even better as an bool). 
This may be the reason why the code only works with less than 50 frames/s. 

My guessing is that testing if new frames are available using interrupt is faster than using `CAN.checkReceive()`.

zentT, 2020-07-24:
This text is on top of most of the examples available with this library. That's the reason why I think it could be important to understand it / describe it better. 

`My guessing is that testing if new frames are available using interrupt is faster than using CAN.checkReceive().

`
It's strange because to me this statement says otherwise : 
=> When in interrupt mode, the data coming can't be too fast, must >20ms, or else you can use check mode

I have indeed declared volatile bool flagRecv = 0. 

It's working like a charm : with a master talking to 2 slaves and taking back information from them. 

But I have a big issue with the last problem submitted here #99. I'm using filters and mask so I don't have an interrupt every message sent to any node. I can't understand why the filters won't work properly when the USB is disconnected from the Arduino Nano. 

Thank you for your answer.

closed 2020-12-10 by Pillar1989