Seeed-Studio/Seeed_Arduino_CAN #105: Arduino DUE

zentT, 2020-12-23:
Hi, I'd like to use this library with a Arduino DUE, communication via CAN with a Arduino UNO. 
Could you give me hints on what to change to make this library compatible with DUE ? Thanks a lot,

Térence

Pillar1989, 2020-12-24:
It should be compatible now. Have you tested it? Did you encounter any problems?

Wodie, 2021-01-03:
Hi, I am testing and it does not work.

Pillar1989, 2021-01-04:
I just updated the code, please try again.

zentT, 2021-01-05:
Hi, I'm trying to use your library with an Arduino Due (84MHz) and a
MCP2515 but I can't make it work, would you have any advice on what to do ?

Thanks,

Térence

>

-- 
 
<https://iqoya-connect.digigram.com/?utm_source=IQOYA%20CONNECT&utm_medium=email%20signature%20Digigram&utm_campaign=email%20signature%20Digigram> 
  <https://youtu.be/fvNhq0lGXVA>

Wodie, 2021-01-05:
Hi, no luck even with the examples. Seems your SoftwareSerial.h library is now deprecated I cpild not find it and make an example of your code compile.

In file included from /var/folders/fv/9mkmvhqs6lg6tgfdk5qwpjlm0000gn/T/arduino_modified_sketch_886409/canbus-monitor.ino:16:0:
can-serial.h:24:28: fatal error: SoftwareSerial.h: No such file or directory
 #include "SoftwareSerial.h"
                            ^
compilation terminated.
exit status 1
SoftwareSerial.h: No such file or directory

Saludos,
IEC Juan Carlos Pérez De Castro
KM4NNO

WLPS
Wodielite Production Services
Mexico City,
Mexico
Mob: +52 (55)54356002
www.wodielite.com

To see the privacy statement visit: www.wodielite.com

GunZi200, 2021-01-05:
I don't think SoftwareSerial supports Arduino Due. Is there a good reason not to use Hardware Serial?

Wodie, 2021-01-05:
Sorry, how can I change it to use Hardware Serial and compile properly?

GunZi200, 2021-01-06:
I created a pull request here: https://github.com/Seeed-Studio/Seeed_Arduino_CAN/pull/108 I don't have an Arduino DUE to test, so please let me know if this doesn't work :)

zentT, 2021-01-08:
Working with a Due and a MCP2515 + TJA1051, it does work with your library. Thank you very much

PR #108 (2021-01-08): Use HardwareSerial instead of SoftwareSerial in CAN-monitor example
I created this to help resolve #105 

I do not have an Arduino DUE to test but this does build successfully. Someone with an Arduino DUE will need to verify this actually works.
closed 2021-01-08 by zentT