Seeed-Studio/Seeed_Arduino_CAN #106: Getting error while compiling through arduino IDE

mjat2020, 2021-01-03:
**Describe the bug**
cannot declare variable 'CAN' to be of abstract type 'MCP_CAN' 

I got this error while compiling through arduino IDE while declaring the class object by this header file

Pillar1989, 2021-01-04:
We have recently updated our code and I would suggest modifying it to accommodate the latest software. If you are not willing to modify anything, I suggest you are using this branch of the code.
https://github.com/Seeed-Studio/Seeed_Arduino_CAN/tree/old

closed 2021-01-04 by Pillar1989

Ipsitakoley, 2021-10-04:
@Pillar1989 Can you please provide the link where we can get the latest library and sample code to accommodate the changes?

Pillar1989, 2021-10-08:
@Ipsitakoley  `master` branch is the latest library.