Seeed-Studio/Seeed_Arduino_CAN #109: No error handling in mcp2515_read_canMsg and *len parameter is redundant

GunZi200, 2021-01-10:
Hello I've been using your library since before v2.0.0 and I am still using v1.0.0 though I see the bug I want to report is still relevant.

**Describe the bug**

In the function `mcp2515_read_canMsg` the parameter `volatile byte* len` is never used and this is dangerous. The user can input whatever he wants but later the value will be assigned 

`*len = pMsgSize & MCP_DLC_MASK;`

I propose to add a check after this line to verify if the supplied `volatile byte* buf` has `buffer_length >= *len`, if not then `return` with an error because overflow will occur. This took me a very long time for me to debug on an Arduino.

If `volatile byte* buf` has length 5 and `*len` is assigned 8 (by reading 8 from DLC bits) then overflow occurs in this loop:

```cpp
for (i = 0; i < *len && i < CAN_MAX_CHAR_IN_MESSAGE; i++) {
    buf[i] = spi_read();
}
```

There is no guarantee that buf will be able to fit the data and the user will be forced to use always a buffer with length 8.

Also what use *len has in seems redundant aswell.
```cpp
byte mcp2515_can::readMsgBufID(byte status, volatile unsigned long* id, volatile byte* ext, volatile byte* rtrBit,
                           volatile byte* len, volatile byte* buf)
```

I think here we would want `mcp2515_read_canMsg` to return `CAN_OK` directly.

```cpp
if (status & MCP_RX0IF) {                                        // Msg in Buffer 0
    mcp2515_read_canMsg(MCP_READ_RX0, id, ext, rtrBit, len, buf);
    rc = CAN_OK;
} else if (status & MCP_RX1IF) {                                 // Msg in Buffer 1
    mcp2515_read_canMsg(MCP_READ_RX1, id, ext, rtrBit, len, buf);
    rc = CAN_OK;
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Transmit a CAN message with 8 bytes to MCP2515 or MCP2518FD
2. Declare `byte buf[5] = {0}` and call `readMsgBuf(*len, buf)`
3. Overflow occurs as library attempts to store 8 bytes in 5 byte array.

**Expected behavior**
No overflow occurs and either the library reads only until `buf` is full or returns an error to indicate the `buf` array is too small.

**Screenshots**
```c
void mcp2515_can::mcp2515_read_canMsg(const byte buffer_load_addr, volatile unsigned long* id, volatile byte* ext,
                                  volatile byte* rtrBit, volatile byte* len, volatile byte* buf) {      /* read can msg                 */
    byte tbufdata[4];
    byte i;

    MCP2515_SELECT();
    /*
    Code between here is not relevant and does not use *len
   */

    byte pMsgSize = spi_read();
    *len = pMsgSize & MCP_DLC_MASK; // <-- len is assigned value from DLC and function input is not used
    *rtrBit = (pMsgSize & MCP_RTR_MASK) ? 1 : 0;
    for (i = 0; i < *len && i < CAN_MAX_CHAR_IN_MESSAGE; i++) {
        buf[i] = spi_read();  // <-- Risk of overflow
    }

    MCP2515_UNSELECT();
}
```

GunZi200, 2021-01-11:
I've updated the issue title and description.

turmary, 2021-01-14:
Obviously, it's a bug ...

turmary, 2021-01-14:
For a temperary solution, please define a buffer at least 8 bytes.

closed 2021-01-29 by Pillar1989