Seeed-Studio/Seeed_Arduino_CAN #111: Extended ID with 0x0..

sundar19, 2021-01-14:
Hi, 
I am currently using the old version of this library .
To send a CAN message I use 
`CAN.sendMsgBuf(0x018c0ef4,1, 8, stmp);` 
I use 29 bit identifier address and have used given 1 for extended ID . But on the receiver side I receive it as 0x18c0ef4 and not 0x018c0ef4 , I think in my case that starting 0 is also required , can you please let me know how to make the ID padded with a 0 before it. 
Thank you

wdross, 2021-01-15:
How the receiver formats the numeric ID is simply a display issue. For example if the data were printed via fprint with "0x%08d" you'd get the zero prefix you think is missing whereas "0x%d" would be what you are seeing now.

sundar19, 2021-01-15:
So this means that the sendMsgBuf() is sending with a padded 0 and it does not strip the 0 off? only the receiver displays it without 0 if I am right?

turmary, 2021-01-15:
@sundar19 yes, that's only a display problem.

also, in C language, 0x1 = 0x01 = 0x001 = 0x0001

sundar19, 2021-01-15:
Yes sir , I am trying to communicate with a battery management system using CAN , They have given a list of identifiers out of which first one is 0x018c0ef4. And the rest of identifiers also starts with 0x0... why should they give an identifier starting with 0x0 ? So all the hex addressing should start with 0x and some value instead of 0 right? ( apologize with me if its out of scope)

sundar19, 2021-01-15:
Sir,
is this library suitable for sae j1939 standard?

wdross, 2021-01-15:
This library is good for sending and receiving CAN messages. J1939 is a protocol, which defines what the bytes within the message (and the COBID) mean.
This library is the lowest level and is simply a way to send or receive a message.

sundar19, 2021-01-15:
Thank you so much sir

closed 2021-01-29 by Pillar1989