Seeed-Studio/Seeed_Arduino_CAN #113: CAN Interrupt Read Faulty

georgiboby, 2021-02-11:
**Describe the bug**
Interrupt Reading of MCP2515 with Arduino doesn't work reliably. 

**To Reproduce**
Try overloading the Bus and receive at one End. After a few messages, the Arduino Will not go into the ISR as the Interrupt Pin always remains low. Thus the Arduino will not read any more messages.

**Reason**
This is because MCP2515 interrupt doesn't work as a trigger interrupt and stays low as long as there is a message in the Buffer. However arduino expects a Pulse everytime a message is added to the Buffer.

The Correct way is to poll the MCP2515 Interrupt Pin in loop and whenever it is LOW, Read the CAN Messages.

Pillar1989, 2021-02-19:
@georgiboby 
@turmary  is in the middle of the Chinese New Year holiday, and he will deal with your questions the first time he returns to the office. Thank you very much for waiting.

turmary, 2021-02-22:
hi @georgiboby 
I tested the example examples/receive_interrupt/receive_interrupt.ino by following your reproduce steps on Arduino UNO,
But not found the stop of reading messages.
Did you modified the example code in your test?

closed 2021-04-21 by Pillar1989