Seeed-Studio/Seeed_Arduino_CAN #114: RTR and EXT

fkaraokur, 2021-02-18:
I have 2 questions!

1. How do we set 1000 kbps and 8 mhz?

2. How do I include the code 0 for rtr and 1 for ext into the id. Can you write sample code?

Pillar1989, 2021-02-19:
Hi @fkaraokur 
@turmary  is in the middle of the Chinese New Year holiday, and he will deal with your questions the first time he returns to the office. Thank you very much for waiting.

turmary, 2021-02-19:
Hi @fkaraokur 
Please refer the examples code examples/sendFD/sendFD.ino.
Question 1:
replace CAN.begin argument CAN_500K_1M with CANFD::BITRATE(1000000UL, 8)
Question 2: 
Use the overloaded version sendMsgBuf(),
```
byte sendMsgBuf(unsigned long id, byte ext, byte rtr, byte dlc,
                          const byte *buf, bool wait_sent = true);
```
for example, 
``
id = 0x12345678UL;
sendMsgBuf(id, 1, 0, dlc, buf);
```

closed 2021-04-21 by Pillar1989