Seeed-Studio/Seeed_Arduino_CAN #115: Can receive normal CAN frames, but not extended ones (ext parameter set)

dhopf, 2021-04-21:
I already commented underneath closed issue #39, but it seems better to be listed as a new issue.
Even with the newest version of the library I'm unable to receive any CAN-frames which have been sent by another CAN-Shield with the 'ext' bit being set to 1 in the sendMsgBuf function.

***My setup***
- Teensy 3.2 Board
- 2x MCP2515 CAN-Shields (I use the ones from [AZ Delivery](https://www.az-delivery.de/en/products/mcp2515-can-bus-modul) featuring an 8MHz crystals)
- Both CAN-Shields are connected to SPI, both using different CS pins
- CAN-connection between both CAN-Shields, 120 Ohm termination resistors bridged on every Shield
- Everything is wired up on a breadboard

***The issue***
I took the "send_receive" example and slightly modified it (slower sending, better error checking/reporting, adjust to 250kbps & 8MHz crystal). I get _"CAN BUS sendMsgBuf ok!"_ printed for every frame sent, but `CAN_RECEIVE.checkReceive()` always returns `CAN_NOMSG`.
Following is the code for sending extended CAN-Frames:
```
/*
    This is an example of sending and receiving on its own, connect two channels to each other using jumpers.
*/

#include <SPI.h>
#include "mcp2515_can.h"

/*SAMD core*/
#ifdef ARDUINO_SAMD_VARIANT_COMPLIANCE
    #define SERIAL SerialUSB
#else
    #define SERIAL Serial
#endif

const int SPI_CS_PIN_SEND = 15;
const int SPI_CS_PIN_RECEIVE = 10;

mcp2515_can CAN_SEND(SPI_CS_PIN_SEND);
mcp2515_can CAN_RECEIVE(SPI_CS_PIN_RECEIVE);

unsigned char len = 0;
unsigned char buf[8];

void setup() {
    SERIAL.begin(115200);
    while(!Serial); // wait for Serial

    pinMode(SPI_CS_PIN_SEND,OUTPUT);
    digitalWrite(SPI_CS_PIN_SEND, HIGH);
    pinMode(SPI_CS_PIN_RECEIVE,OUTPUT);
    digitalWrite(SPI_CS_PIN_RECEIVE, HIGH);
    delay(1);

    if (CAN_SEND.begin((byte)CAN_250KBPS, MCP_8MHz) != 0 || CAN_RECEIVE.begin((byte)CAN_250KBPS, MCP_8MHz) != 0) {
      Serial.println("CAN-BUS initiliased error!");
      while(1);
    }

    SERIAL.println("CAN BUS Shield init ok!");
}

unsigned char stmp[8] = {0, 0, 0, 0, 0, 0, 0, 0};
void loop() {
    // send data:  id = 0x1, extended frame, data len = 8, stmp: data buf
    stmp[7] = stmp[7] + 1;
    if (stmp[7] == 100) {
        stmp[7] = 0;
        stmp[6] = stmp[6] + 1;

        if (stmp[6] == 100) {
            stmp[6] = 0;
            stmp[5] = stmp[6] + 1;
        }
    }

    byte ret;
    if (CAN_OK == (ret = CAN_SEND.sendMsgBuf(0x1, 1, sizeof(stmp), stmp))) {
        SERIAL.println("CAN BUS sendMsgBuf ok!");
    } else {
        SERIAL.print("CAN BUS sendMsgBuf ERROR: ");
        SERIAL.println(ret);
    }
    delay(10);                       // Wait for propagation of frame

    // ---------------------

    if (CAN_MSGAVAIL == CAN_RECEIVE.checkReceive()) {
    // read data,  len: data length, buf: data buf
        SERIAL.println("checkReceive");
        CAN_RECEIVE.readMsgBuf(&len, buf);
        // print the data
        for (int i = 0; i < len; i++) {
            SERIAL.print(buf[i]); SERIAL.print(" ");
        }
        SERIAL.println();
    }
    SERIAL.println("---LOOP END---");
    delay(490); // in total we have >500ms delay --> send twice per second
}

// END FILE
```
If I just change the 'ext' flag to 0 (send standard 11-Bit Identifier CAN frames instead of 29-Bit extended ones), the other CAN shield successfully receives every frame:
`    if (CAN_OK == (ret = CAN_SEND.sendMsgBuf(0x1, 0, sizeof(stmp), stmp))) {`

I really would appreciate if anyone has the opportunity to reproduce the setup and confirm or report different results.

dhopf, 2021-04-22:
I found a solution (thanks to the amazon reviewer mentioning this for one of the CAN-Shield clones):
It seems that the MCP2515 by default filters out extended IDs. I added filter masks to my setup() routine to accept everything (including extended IDs) and now my CAN_RECEIVE gets the frames in both cases: standard and extended identifiers.

```
void setup() {
    SERIAL.begin(115200);
    while(!Serial); // wait for Serial

    pinMode(SPI_CS_PIN_SEND,OUTPUT);
    digitalWrite(SPI_CS_PIN_SEND, HIGH);
    pinMode(SPI_CS_PIN_RECEIVE,OUTPUT);
    digitalWrite(SPI_CS_PIN_RECEIVE, HIGH);
    delay(1);

    if (CAN_SEND.begin((byte)CAN_250KBPS, MCP_8MHz) != 0 || CAN_RECEIVE.begin((byte)CAN_250KBPS, MCP_8MHz) != 0) {
      Serial.println("CAN-BUS initiliased error!");
      while(1);
    }

    // Set filter masks to accept all frames, including extended ones
    CAN_RECEIVE.init_Mask(0, CAN_STDID, 0); // Mask no Standard-ID bits to be filtered (0x00)
    CAN_RECEIVE.init_Mask(1, CAN_EXTID, 0); // Mask no Extended-ID bits to be filtered (0x00)
    CAN_RECEIVE.init_Filt(0, CAN_STDID, 0); // Init filter #0.0 in mcp2515 for no filtering
    CAN_RECEIVE.init_Filt(1, CAN_STDID, 0); // Init filter #0.1 in mcp2515 for no filtering
    CAN_RECEIVE.init_Filt(2, CAN_EXTID, 0); // Init filter #1.2 in mcp2515 for no filtering
    CAN_RECEIVE.init_Filt(3, CAN_EXTID, 0); // Init filter #1.3 in mcp2515 for no filtering
    CAN_RECEIVE.init_Filt(4, CAN_EXTID, 0); // Init filter #1.4 in mcp2515 for no filtering
    CAN_RECEIVE.init_Filt(5, CAN_EXTID, 0); // Init filter #1.5 in mcp2515 for no filtering

    SERIAL.println("CAN BUS Shield init ok!");
}
```
Can the developers confirm that this is intentional, or should the lib automatically set these filters on object creation, such that extended IDs are not filtered by default?

Pillar1989, 2021-05-18:
This is a generic library and it does not automatically filter out this key information

closed 2021-05-18 by Pillar1989