Seeed-Studio/Seeed_Arduino_CAN #116: MCP2518FD Filters not working

TyphoonSB, 2021-04-29:
Hi,

I am trying to get the Filter function working......

I understood that the MCP2518FD has 32 Filters, and each filter has a mask and a filter value...

CAN.init_Mask(0, 0, 0x7FF);
CAN.init_Filter(0, 0, 0x300);
...
CAN.init_Mask(31, 0, 0x7FF);
CAN.init_Filter(31, 0, 0x301);
etc.

Has someone an idea or some sample code for me?

Thanks in advance...

Simon

Pillar1989, 2021-05-10:
https://github.com/Seeed-Studio/Seeed_Arduino_CAN/blob/master/examples/set_mask_filter_recv/set_mask_filter_recv.ino @TyphoonSB  Please refer this example

TyphoonSB, 2021-05-10:
Thanks for feedback, but just this example was not working....

in the meantime I found out that it was an error in the libary....  after changing the filter / mask you have to enable it by swoping the FILTEN bit from 0 to 1.

Then it works fine. I created a pull request, and it should already here in the libary.

closed 2021-05-10 by TyphoonSB