Seeed-Studio/Seeed_Arduino_CAN #135: Problem with CAN BUS Receive Data

Shoihya, 2022-03-30:
Hello everyone,

i am trying to test Recieve and send signal using Seed-Arduino CAN Shield V2.0 using arduino Mega 2560 the sending signal working good actualy. but i have issue with recieve data:

The problem is that the code cannot get the data length and always get it zero so i cannot got the write signal as i need it to programm some digital stuff.

Aslo the Loop didnot want to work until i put (CAN_MSGAVAIL != CAN.checkReceive())
insteadt (CAN_MSGAVAIL = CAN.checkReceive()) . I have no idea why !

What i understand from the syntax example code that:
1- len=0 is a defined var will be filled as soon as the data length will delivered then gonna use this data to read it or control something else.

2- [[ CAN_MSGAVAIL == CAN.checkReceive() ]] this line is for checking if data is comming and will not be equal until data is comming, howerver if i send data throw CAN.
Actually (CAN_MSGAVAIL != CAN.checkReceive()) working and allow the loop to continiue.  but the further Loop (for Loop) didn't work as "len" always equal zero cause it didn't get the new value.

I am testing the send and recieve using CANoe Software and i only can see the data that i send, but i cannot read this data back to my Arduino Program.

Questions:

Is there anyway to check/Solve that or should i add something in the code.

I hope my attached code lead u to find the issue.

Thanks 
Ahmed

[image]
[Reciev_Send_Can_Test.zip](https://github.com/Seeed-Studio/Seeed_Arduino_CAN/files/8377734/Reciev_Send_Can_Test.zip)

lakshanthad, 2022-04-07:
Hello @Shoihya,

1. Have you properly referred to our wiki at first to set up the CAN-BUS Shield?
https://wiki.seeedstudio.com/CAN-BUS_Shield_V2.0

2. Could you please share with me the hardware diagram of your current setup?

Thank you.

mattewhite, 2022-04-08:
Hello, 

I am experiencing the same issue. 
I have reviewed the wiki for setup of the CAN-BUS shield and cannot seem to find an answer to my question.

Hardware is setup as follows:
Arduino Uno board with CAN hat from seeed studio(COM1). DB9 connected directly to "PEAK CAN" usb CAN output device(COM3). 

I am generating a CAN signal from Matlab Simulink which I am attempting to read via the Arduino.

Using CAN-bus shield library
Using Receive-blink example

If I leave this line as is [[ CAN_MSGAVAIL == CAN.checkReceive() ]]
       Then the serial port monitor displays:

                CAN init ok!
                ?xx?????xxxxx??????xxxx?x?x??

If I change the line to [[ CAN_MSGAVAIL != CAN.checkReceive() ]]
       Then the serial port monitor displays:

              CAN init ok!
               -------------------------------------------------
              get data from ID: 0x
              0
               -----------------------------
               g?x???xx?x?x?x?x???x?x???xxx?x?xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

This is all weather or not I am not sending any data to the Arduino from the other device. I have also tried this while the arduino is the ONLY device connected to my machine.

any ideas or suggestions are greatly appreciated!

Pillar1989, 2022-04-09:
The inability of the CAN bus to communicate is not a one-sided problem. It is clear that after so many years of iterations of our software, basic communication is certainly not a problem. If you need to verify our software, please select two identical Seeed CAN-BUS  Shield one for receiving and one for sending to verify. We have no way to support the peripheral you need to communicate with, it is beyond our support. @mattewhite  @lakshanthad  @Shoihya

RufusVS, 2022-09-24:
You should at least check the return value from CAN.checkReceive() to see what the result code is.  That could be a clue to your problem.  The fact that it is not equal to CAN_MSGAVAIL means there's some kind of error or warning status.  The last line of your sample captures looks like your serial port baud rate changed somehow.  Either from your terminal program or your arduino device.

Lesords, 2024-10-16:
Hello,

I'm going to close this issue, feel free to re-open it if you have any other questions.

closed 2024-10-16 by Lesords