Seeed-Studio/Seeed_Arduino_CAN #137: no matching function for call to 'mcp2515_can::sendMsgBuf(int, int, int, unsigned char [8])'

IAmOrion, 2022-10-20:
**Describe the bug**
I opened the "send" example, hit verify, and I'm presented with: 
"no matching function for call to 'mcp2515_can::sendMsgBuf(int, int, int, unsigned char [8])'"

**To Reproduce**
Set BOARD to Adafruit SAMD > Adafruit Feather M0 (SAMD21)

**Expected behavior**
I expected it to verify/compile without issue

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Monterey 12.6
 - Browser: Vivaldi
 - Version: 5.5.2805.38 (Stable channel) (x86_64) 

**Additional context**
Arduino IDE 1.8.19

My Seeed Studio library was outdated so updated it to the latest 2.3.0.  I then wanted to amend my sketch code to work with the updated library.  When I was unable to get it to work, I then opened the various examples included in the library.  They all have the same Error.

The previous versions worked for both my Adafruit Feather M0 BLE (SAMD21), and my Adafruit Feather 32u4 BLE, so it's can't (or shouldn't) be a problem with my chosen board as previously the code compiled fine including the examples, but with the 2.3.3. library, even the examples fail.

Fred7194, 2022-10-23:
I had same issue. Finally I tried it on PlatformIO and it worked. Still trying to figure out issue.

m516, 2022-10-24:
I remember having this issue before, but I couldn't replicate it when compiling `send` for these configurations:

- master branch, CAN_2515 enabled
- 2.3.3 release, CAN_2515 enabled
- master branch, CAN_2518FD enabled
- 2.3.3 release, CAN_2518FD enabled

I might be able to help if you could share a bit more about the hardware and could see the compiler logs. For now, consider calling `sendMsgBuf`  with six-parameters as described [in the source code](https://github.com/Seeed-Studio/Seeed_Arduino_CAN/blob/eb6887ba3c0c0157cc09478230c637e8e11b6be2/src/mcp_can.h#L98)

IAmOrion, 2022-10-24:
I should've returned and updated this -- but yes, previously, the RTRBit was *not* needed, so, for example, could just use:

`CAN.sendMsgBuf(MSG_OUT_ROTARY_INIT, 0, msglength, buf);`

which was basically (ID, EXT, LEN, MESSAGE)

But now, it appears the RTRBit is required, so It's effectively

`CAN.sendMsgBuf(MSG_OUT_ROTARY_INIT, 0, 0, msglength, buf);`

Which is (ID, EXT, RTR, LEN, MESSAGE) ... generally speaking, (at least for most basic stuff) the RTR bit would be 0

I will close this as mine is all working now once I sussed that out

closed 2022-10-24 by IAmOrion