Seeed-Studio/Seeed_Arduino_CAN #140: MCP2515 - Inverted frame sending

andrejrcarvalho, 2023-03-08:
**Bug description**
I'm using your library to send CAN frames to a network that implements the UAVCAN stack. In this stack, the frames should be sent in a correct order to be possible to decode the full message.
When the network frequency is low, like 10KBs, the 3 buffers of the MCP2515 are used to send the frames. This lead to an inverted sent of the frames.

**Example:**
I have 3 frames to send, and the frames are added to the queue with the following order:
1. Frame 0
2. Frame 1
3. Frame 2

In the MCP2515, the buffers are populated with that frames and should lock like this:

| Buffer 0 | Buffer 1 | Buffer 2 |
|----------|----------|----------|
|  Frame 0 |  Frame 1 |  Frame 2 |

When the MCP2515 sends the frames, the output is inverted. According to the datasheet the MCP2515 sends first, the frame that are in the buffer with the higher number. Following that logic, the output should look like this: 

1. Frame 2
2. Frame 1
3. Frame 0

To prove this, I have made a test with 2 nodes, one sending the messages and other receiving then.

The network is working at 10KBs and to analyze the message I have used the [Drone CAN GUI tool](https://dronecan.github.io/GUI_Tool/Overview/)

The output is this:

[image]

And should be like this:

[image]

To make the result above I have inferted the for loop used to get the next available buffer:

```C
byte mcp2515_can::mcp2515_getNextFreeTXBuf(byte* txbuf_n) {
    byte status = mcp2515_readStatus() & MCP_STAT_TX_PENDING_MASK;
    byte i;

    *txbuf_n = 0x00;

    if (status == MCP_STAT_TX_PENDING_MASK) {
        return MCP_ALLTXBUSY; 
    }

    for (i = MCP_N_TXBUFFERS - nReservedTx; i >= 0; i--) {//<<<<<----------  THIS LINE ( before iterate from 0 to 3 and now iterate from 3 to 0)
        if ((status & txStatusPendingFlag(i)) == 0) {
            *txbuf_n = txCtrlReg(i) + 1;
            mcp2515_modifyRegister(MCP_CANINTF, txIfFlag(i), 0);
            return MCP2515_OK;
        }
    }

    return MCP_ALLTXBUSY;
}
```

techie66, 2023-08-03:
PR?

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

We welcome you to submit PR.

Lesords, 2024-11-04:
Hello,

I\'m going to close this issue, feel free to re-open it if you have any other questions.

[Autocommentary]

closed 2024-11-04 by Lesords

techie66, 2025-02-02:
Sorry it took so long for me to get to this. I just made a simple PR based off the above comments. It seems to work just as well, but now in proper order if frames are added faster than they can be sent.