Seeed-Studio/Seeed_Arduino_CAN #147: Filters and masks do not seem to work properly

reldneb, 2023-08-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

DevZombi, 2024-05-09:
I am having this same exact issue. I can Mask and Filter extended IDs but can't filter standard IDs.

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

reldneb, 2024-10-16:
Hello, I already finished my CAN project a few months ago and removed 12 Arduino boards from my motorhome.
The Can-Bus could not be filtered properly.
I have given up on this and have de-digitised the vehicle again. It took a lot of time and money but I still enjoyed it.

Thanks for the message

Lesords, 2024-10-16:
OK, I understand.

Apologize again.

closed 2024-11-04 by Lesords