Seeed-Studio/Seeed_Arduino_CAN #149: CAN.getCanId() gets the ID of the TX MSG insted of the RX MSG

Damians99, 2023-12-21:
I send a message periodically (100 Hz) with the CAN-SHIELD (MPC2515).
In parallel, I execute the following function:

```
while (CAN_MSGAVAIL == CAN.checkReceive())
    {
        CanRxInterrupt();
        //Serial.println("MSG detects on RX");
    }

void CanRxInterrupt() {

    unsigned int canId = CAN.getCanId();
    unsigned char len;
    unsigned char buf[8];
    CAN.readMsgBuf(&len, buf);

    Serial.println(canId, HEX);
```

`CAN_MSGAVAIL == CAN.checkReceive()  `

works perfectly with an incoming message and CanRxInterrupt() is called when a MSG is incomming. However, the following line:

`unsigned int canId = CAN.getCanId()`

always returns the ID of the just periodically sent message and not that of the incoming message. Sommetimes when the incomming MSG ist sent with a verry high frequence i got the correct ID, but nearly never.

 - OS: Windows 11, Arduino, VS Code

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

Damians99, 2024-10-16:
Hii, thanks for responding.
No i was able to solve it by myself while checking the souce code. The Problem was caused by me.

Problem/Solution is:
CAN.readMsgBuf() has to be executed before I execute the CAN.getCanID()

Otherwise the stored Can ID will be the one i just sent and not the one i just recived(because i never read the RX Buffer)

Now it seems logical, but i didnt know that the stored ID want be updated with the execution of CAN.getCanID()

Thanks 
Best Regards 
Damian

Lesords, 2024-10-17:
Okay, with that I'm going to close this issue, if you have any other questions you're welcome to reopen the issue.

closed 2024-10-17 by Lesords