Seeed-Studio/Seeed_Arduino_CAN #155: INT declaration when no interrupts used?

mikechabnl, 2025-05-28:
I'm using a setup of two Arduino Uno with a seeed studio CAN-BUS SHIELDs. When I use the default receive and send programs, I will often see the systems' INT LED light up. This LED will light up even if there is not interrupt being used. This LED will also light up if I switch the jumpers on the back of the CAN-BUS SHIELD to pin 3 instead of 2, even when I do not initialize an ISR. 

Is there any function for the INT signal when not using interrupts? It does not look like I can even set up the interrupt pin in either program as you can set the CS pin. 

For completeness, the send and receive programs are below.

Send
```
// demo: CAN-BUS Shield, send data
// loovee@seeed.cc

#include <SPI.h>

#define CAN_2515
// #define CAN_2518FD

// Set SPI CS Pin according to your hardware

#if defined(SEEED_WIO_TERMINAL) && defined(CAN_2518FD)
// For Wio Terminal w/ MCP2518FD RPi Hat：
// Channel 0 SPI_CS Pin: BCM 8
// Channel 1 SPI_CS Pin: BCM 7
// Interupt Pin: BCM25
const int SPI_CS_PIN  = BCM8;
const int CAN_INT_PIN = BCM25;
#else

// For Arduino MCP2515 Hat:
// the cs pin of the version after v1.1 is default to D9
// v0.9b and v1.0 is default D10
const int SPI_CS_PIN = 9;
const int CAN_INT_PIN = 2;
#endif

#ifdef CAN_2518FD
#include "mcp2518fd_can.h"
mcp2518fd CAN(SPI_CS_PIN); // Set CS pin
#endif

#ifdef CAN_2515
#include "mcp2515_can.h"
mcp2515_can CAN(SPI_CS_PIN); // Set CS pin
#endif

void setup() {
    SERIAL_PORT_MONITOR.begin(115200);
    while(!Serial){};

    while (CAN_OK != CAN.begin(CAN_500KBPS)) {             // init can bus : baudrate = 500k
        SERIAL_PORT_MONITOR.println("CAN init fail, retry...");
        delay(100);
    }
    SERIAL_PORT_MONITOR.println("CAN init ok!");
}

unsigned char stmp[8] = {0, 0, 0, 0, 0, 0, 0, 0};
void loop() {
    // send data:  id = 0x00, standrad frame, data len = 8, stmp: data buf
    stmp[7] = stmp[7] + 1;
    if (stmp[7] == 100) {
        stmp[7] = 0;
        stmp[6] = stmp[6] + 1;

        if (stmp[6] == 100) {
            stmp[6] = 0;
            stmp[5] = stmp[5] + 1;
        }
    }

    CAN.sendMsgBuf(0x00, 0, 8, stmp);
    delay(100);                       // send data per 100ms
    SERIAL_PORT_MONITOR.println("CAN BUS sendMsgBuf ok!");
}

// END FILE

```

Receive
```
// demo: CAN-BUS Shield, receive data with check mode
// send data coming to fast, such as less than 10ms, you can use this way
// loovee, 2014-6-13
#include <SPI.h>

#define CAN_2515
// #define CAN_2518FD

// Set SPI CS Pin according to your hardware

#if defined(SEEED_WIO_TERMINAL) && defined(CAN_2518FD)
// For Wio Terminal w/ MCP2518FD RPi Hat：
// Channel 0 SPI_CS Pin: BCM 8
// Channel 1 SPI_CS Pin: BCM 7
// Interupt Pin: BCM25
const int SPI_CS_PIN  = BCM8;
const int CAN_INT_PIN = BCM25;
#else

// For Arduino MCP2515 Hat:
// the cs pin of the version after v1.1 is default to D9
// v0.9b and v1.0 is default D10
const int SPI_CS_PIN = 9;
const int CAN_INT_PIN = 2;
#endif

#ifdef CAN_2518FD
#include "mcp2518fd_can.h"
mcp2518fd CAN(SPI_CS_PIN); // Set CS pin
#endif

#ifdef CAN_2515
#include "mcp2515_can.h"
mcp2515_can CAN(SPI_CS_PIN); // Set CS pin
#endif                           

void setup() {
    SERIAL_PORT_MONITOR.begin(115200);

    while (CAN_OK != CAN.begin(CAN_500KBPS)) {             // init can bus : baudrate = 500k
        SERIAL_PORT_MONITOR.println("CAN init fail, retry...");
        delay(100);
    }
    SERIAL_PORT_MONITOR.println("CAN init ok!");
}

void loop() {
    unsigned char len = 0;
    unsigned char buf[8];

    if (CAN_MSGAVAIL == CAN.checkReceive()) {         // check if data coming
        CAN.readMsgBuf(&len, buf);    // read data,  len: data length, buf: data buf

        unsigned long canId = CAN.getCanId();

        SERIAL_PORT_MONITOR.println("-----------------------------");
        SERIAL_PORT_MONITOR.print("Get data from ID: 0x");
        SERIAL_PORT_MONITOR.println(canId, HEX);

        for (int i = 0; i < len; i++) { // print the data
            SERIAL_PORT_MONITOR.print(buf[i], HEX);
            SERIAL_PORT_MONITOR.print("\t");
        }
        SERIAL_PORT_MONITOR.println();
    }
}

/*********************************************************************************************************
    END FILE
*********************************************************************************************************/

```

Yuan-AZKKXL, 2025-06-03:
The state of the INT pin is automatically controlled by the CAN controller and is independent of whether the Arduino processes interrupts. Even if you haven’t written interrupt handling code, the CAN controller will still pull the INT pin low when an event occurs.

Second, I haven’t reproduced the issue you described. Since I only have one Arduino Uno with a Seeed Studio CAN-BUS Shield, I used it as the receiving device with your receive code. For the sending device, I used a Xiao ESP32-C3 (https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/). In my test, the RX and INT LEDs only lit up when data was received. In the picture below, I unplugged the CAN_L wire to disconnect the two devices, and the INT LED did not light up. Could you please point out if there is an error in my setup so that I can reproduce your issue?

[image]

wdross, 2025-06-06:
Another way to look at using (or not) the INT signal is: when will your code look for the CAN message? 

If not serviced by an interrupt, then your program will have to poll and keep asking "got a message?" of the hardware, which (depending on any other tasks your program is doing) could take "some time" to get the message.  The CAN hardware can store a message or two, but if your program doesn't get the message quick enough, the next message will be missed once the buffer is full.  At 500KBPS a message can be received in 1/4ms, so if your program takes a 10ms nap it's possible up to 40 messages can go by, depending on what's all on the bus

Yuan-AZKKXL, 2025-06-09:
I understand your concern now. Let me clarify this with the help of the schematic diagram (https://files.seeedstudio.com/wiki/CAN-BUS-Shield-V2.0/res/CAN-BUS%20Shield%20%20v2.0.pdf). In the CAN-BUS SHIELD design, the MCP2515 chip handles CAN message transmission and reception. When it receives a message, the MCP2515 will automatically pull the INT pin low. The MCU can then capture this signal to trigger an interrupt - the LED lights up because of the INT pin being pulled low, not because the interrupt service routine (ISR) is executed.

To elaborate, once the MCP2515 pulls the INT pin low:

1.You can connect this pin to an LED to visually indicate message reception.
2.You can also connect it to an MCU pin to enable interrupt - based message handling.

As @wdross mentioned, if you need to process every CAN message in real - time (without missing any), it is recommended to interact with the INT pin. However, if you choose to handle messages via polling (e.g., using CAN.checkReceive()), direct interaction with the INT pin is not mandatory.

My personal recommendation is to connect the INT pin in your PCB design. Even if you don't use interrupts now, future users or applications might need this functionality. Leaving this option available makes your design more flexible and user - friendly.